mmarchetti/DirectIO #2: ARM-based Arduinos (e.g. Arduino Due) are not supported

mmarchetti, 2015-01-17:
The current port mapping scheme doesn't work on Arduino ARM variants (currently the Due board). The Arduino ARM libraries do a lot of work per I/O transaction, some of which can be moved into initialization as is done here. Also there are clearly opportunities for inlining, which not only eliminates function call overhead but can often eliminate one branch of an if/else entirely by evaluating the condition at compile time.

JohnLinford, 2016-03-06:
I'm messing with an Arduino Due app that could certainly use this DirectIO library. Do you have any plans to add support in due(!) course? Thanks.

mmarchetti, 2016-03-07:
Hi John, I haven't looked at the Due support in a while - it may be some time before I get back to it. It was a bit more complex than the AVR case, but I think there is an opportunity for some speed improvement by caching port addresses (like the dynamic pin number case) and calling PIOSet directly.

mmarchetti, 2018-10-12:
#17 should help with this - at least enabling a sketch that uses DirectIO to run on Due (without speedup at this time).

mmarchetti, 2018-10-16:
#18 provides accelerated IO for Due

closed 2018-11-01 by mmarchetti