mmarchetti/DirectIO #7: DirectIO does not compile under Arduino 1.6.6 AVR 1.6.9

arkhipenko, 2015-11-07:
Hi,
DirectIO no longer compiles under the new avr board definitions v1.6.9 released with IDE 1.6.6
Could you please have a look?
Here is the output for the first example:

Arduino: 1.6.6 (Windows 8.1), Board: "Arduino/Genuino Uno"

In file included from C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/DirectIO.h:23:0,

                 from C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO\examples\direct_io\direct_io.ino:1:

C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:56:22: error: field initializer is not constant

 _define_port(PORT_B, PINB, PORTB, DDRB);

                      ^

C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:46:31: note: in definition of macro '_define_port'

         static const u16 in = IN_REG; \

                               ^

C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:56:28: error: field initializer is not constant

 _define_port(PORT_B, PINB, PORTB, DDRB);

                            ^

C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:47:32: note: in definition of macro '_define_port'

         static const u16 out = OUT_REG; \

                                ^

C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:56:35: error: field initializer is not constant

 _define_port(PORT_B, PINB, PORTB, DDRB);

                                   ^

C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:48:32: note: in definition of macro '_define_port'

         static const u16 dir = DIR_REG; \

                                ^

C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:60:22: error: field initializer is not constant

 _define_port(PORT_C, PINC, PORTC, DDRC);

                      ^

C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:46:31: note: in definition of macro '_define_port'

         static const u16 in = IN_REG; \

                               ^

C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:60:28: error: field initializer is not constant

 _define_port(PORT_C, PINC, PORTC, DDRC);

                            ^

C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:47:32: note: in definition of macro '_define_port'

         static const u16 out = OUT_REG; \

                                ^

C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:60:35: error: field initializer is not constant

 _define_port(PORT_C, PINC, PORTC, DDRC);

                                   ^

C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:48:32: note: in definition of macro '_define_port'

         static const u16 dir = DIR_REG; \

                                ^

C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:64:22: error: field initializer is not constant

 _define_port(PORT_D, PIND, PORTD, DDRD);

                      ^

C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:46:31: note: in definition of macro '_define_port'

         static const u16 in = IN_REG; \

                               ^

C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:64:28: error: field initializer is not constant

 _define_port(PORT_D, PIND, PORTD, DDRD);

                            ^

C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:47:32: note: in definition of macro '_define_port'

         static const u16 out = OUT_REG; \

                                ^

C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:64:35: error: field initializer is not constant

 _define_port(PORT_D, PIND, PORTD, DDRD);

                                   ^

C:\Users\arkhi_000\Documents\Arduino\libraries\DirectIO/ports.h:48:32: note: in definition of macro '_define_port'

         static const u16 dir = DIR_REG; \

                                ^

exit status 1
Error compiling.

arkhipenko, 2015-11-07:
Actually I think it does not work with 1.6.6 regardless of the board definitions - same issue with avr 1.6.8 and 1.6.7... Had to downgrade back to IDE 1.6.5 to continue working with DirectIO... not ready to give up this library :)

arkhipenko, 2015-11-16:
Arduino team suggested a patch for ports.h:

https://github.com/arduino/arduino-builder/issues/59

could you please review?
It worked for me for Uno and Nano sketches.

mmarchetti, 2015-11-17:
Yes, I'm OK with the patch. Can you please open a PR here?

arkhipenko, 2015-11-17:
I didn't fork the code, just updated port.h locally...
The link above has the diff. Would you be able to apply it?

mmarchetti, 2015-11-17:
Yes. Thanks for the patch.

arkhipenko, 2015-11-17:
No thanks to me - just a messenger. The Arduino team found the solution.

I love DirectIO and use it in many of my projects (for instance: http://www.instructables.com/id/Arduino-Nano-based-Hexbug-Scarab-Robotic-Spider/)

Thank you for creating it!

commit e41bdd57 (2015-11-19): Fix build issue with Arduino 1.6.6

Fixes #7
closed 2015-11-19 by mmarchetti