mmarchetti/DirectIO #8: Arduino: 1.6.12 (Mac OS X), Board: "Arduino/Genuino Uno"  Specified folder/zip file does not contain a valid library

jonathanmf, 2016-09-30:
Hi,

I try to add this library to IDE 1.6.12 on a Mac but I can't. This message is always coming up. 
Do you know why? This is the first time that I see this warning.

Thx for your help!

Jonathan

mmarchetti, 2016-09-30:
Hi @jonathanmf, the GitHub zip download gives you a snapshot of the entire repository. If you unzip that file, you'll get a folder named "DirectIO-master". Use the "Add .ZIP library" command and choose the DirectIO folder inside the DirectIO-master folder.

I'll keep this issue open and will see if moving the .h files to the main directory here will help.

jonathanmf, 2016-09-30:
Hi ! You right I did that (manually,. One by one).and it works thx a lot !

mmarchetti, 2016-09-30:
Great! I pushed a commit to move the files to the base directory, so now you can download and install the ZIP file from GitHub.

closed 2016-09-30 by mmarchetti