mmarchetti/DirectIO #11: Implementation of DirectIO in other libraries

mathgcosta, 2017-11-17:
I really like to use this DirectIO lib.
What I'm try to do is implement it in other libraries, like <LiquidCrystal.h> for example.
Replace digitalWrite() for Output<>. But I got some issues to implement this in cpp.
Anybody already did that? Create an object Class in other Class.

Thanks.

mmarchetti, 2017-11-17:
Hi @mathgcosta, it should be possible to create DirectIO objects as class members... like this (which I haven't tested):

```
#include <DirectIO.h>

#define CONTROL_PIN 17

class LCD {
  public:
    LCD();
    void DoSomething();
  private:
    Output<CONTROL_PIN> control_pin;
}

LCD::LCD()
{
  // initialization code here
}

void LCD::DoSomething()
{
  this->control_pin = HIGH;
}

```

What problems are you seeing?

mathgcosta, 2017-11-17:
Thanks @mmarchetti, I try to use this way that you propose and works!
The thing is, the LCD class has a constructor that gets the pins number and puts on the private variables.

LCD.h:
[image]

LCD.cpp: [constructor]
[image]

So the CONTROL_PIN cannot be a constant and that I got stuck.
(I put 9 number as example here in my arduino).

mmarchetti, 2017-11-17:
To use pin numbers that aren't specified at compile time, you can use the `InputPin` and `OutputPin` classes instead. Those classes allow you to pass the pin number as a constructor argument. That isn't as fast as the regular DirectIO `Output` class, but still around 3x the speed of digitalWrite.

If speed is essential, you could also consider using constants instead of constructor parameters in your class. That would reduce the flexibility (your users would have to edit your constants for their application) but give you greater speed. You might also look at using the DirectIO `OutputPort` class for your 8-bit data bus, if you can live with the restriction that the pins must be on a single hardware port.

mathgcosta, 2017-11-17:
Thank you so much for help.
I'll have to use OutputPort anyway, I like so much that either.

But, how can I use InputPin allowing pass the pin number as argument. I have tried in some ways but its not work.

My main goal is let LiquidCrystal library more fast, so will need flexibility (I'm going to use interruptions time insted of delay() either). But for now I can use directly the pins number, to solve a problem first.

mmarchetti, 2017-11-17:
You can do something like this:

```
class LCD {
  InputPin my_input;
}

LCD::LCD(u8 pin_number) :
  my_input(pin_number)
{
}
```

mathgcosta, 2017-11-17:
Hmm. So nice! Thanks.

Although I still have some issues with c++, long time that I did not program this language.

The error was:
[image]

In header I declare on private:
[image]

And in constructor:
[image]

mmarchetti, 2017-11-18:
Place the `rs_pin(rs)` as an initializer - before the opening `{` of the constructor as in my example.

mathgcosta, 2017-11-18:
hmmm.
Thank you so much.
It works!

mmarchetti, 2017-11-18:
Glad to help! Stay in touch with any other feedback you have.

closed 2017-11-18 by mmarchetti