mmarchetti/DirectIO #12: OutputLow not implemented

mathgcosta, 2017-11-28:
Hi @mmarchetti,
When I use OutputLow syntax it shows this error message:

error: 'OutputLow' does not name a type

I check on DirecIO.h header and didn't saw any implementation of these.
I tried to implement _OutputLow_ class based on _Output_, and the result was:

**Original** Output:
[image]

I don't know if is in the best implementation, but it works.

**Modified** OutputLow:
[image]

mmarchetti, 2017-11-28:
Thank you. Are you able to open a PR with your implementation?

The only change I would suggest is to negate the return value of `read()`, so it would be `! bitRead(...)`.

mathgcosta, 2017-11-28:
Hmm. Ok.
I can try do a Pull Request, never done before.

Thanks!

mmarchetti, 2017-11-29:
Give it a try, if it doesn’t work out, paste your code in a comment here and I will merge it.

mathgcosta, 2017-11-30:
Hi,

So, I tried to do PR in some ways, but nothing happen.
Is it allowed to do that?
Apparently I can't compare the master branch.

[image]

[image]

[image]

mmarchetti, 2017-11-30:
To open a PR, you'd need to fork the repo into your GitHub account, clone your version locally to your machine, make the changes, push to a branch, then open a PR requesting to merge your branch back into my version of the repository. If you haven't done that before, there is a bit of a learning curve.

If you can just paste in the text of your code here, I will create the PR and merge it in.

mathgcosta, 2017-11-30:
So cool. Got it.

Didn't know I had to fork first.
It's done.

Thanks.

mathgcosta, 2017-12-01:
nice!

closed 2017-12-01 by mathgcosta