mmarchetti/DirectIO #14: Using PORT_D for one pin

mathgcosta, 2017-12-17:
Hi,

I'm using atmega328P-AU standalone (the same used in arduino nano boards) and I would like to know if its possible declare, for example,

Output<2> my_output;

as 

OutputPort<PORT_D, 2> my_output;

Or, declare a output pin passing the AVR PORT name, rather than look up equivalent Arduino pin, as reference:

[image]

Thanks!

mmarchetti, 2017-12-17:
You can try using an OutputPort; I'd be interested to know whether a 1-bit port ends up generating the same code. Or you could add a new board type with the associated port-pin mapping in ports.h, which would allow you to use the normal Input and Output classes.

mathgcosta, 2017-12-18:
Wow, 

Great idea. I'll do that.
Thanks a lot.

mathgcosta, 2017-12-19:
Hey,

I did this implementation but something is going wrong (the led doesn't blinked and there wasn't compilation error)

Here is your original port.h:
[image]

And here is what I wrote below:
[image]

Can you see some wrong with it?

mmarchetti, 2017-12-20:
That looks reasonable, provided it matches your board pinout and you have wired the correct pin.

mathgcosta, 2017-12-20:
Yeah,

But doesn't work any output pin on the board, neither blink led nor something else =/
And all wires are correct. The same code using Arduino pins works, that new implementation don't.

mmarchetti, 2017-12-20:
Just checking, is ARDUINO_AVR_NANO defined?

mathgcosta, 2017-12-20:
You mean, when I compile the code? Like here:

[image]

On ports.h file I just removed 
  defined (ARDUINO_AVR_NANO) || \\
from here
[image]

and added a new #elif defined (ARDUINO_AVR_NANO) statement, as I posted before.

mmarchetti, 2017-12-20:
OK, that seems right... just wanted to be sure it was actually defined at compile time so it would use the correct definitions. (you can always verify by putting a `#warning` statement there). And your code is using pin 32 (`Output<32> my_output`)?

mathgcosta, 2018-01-21:
Hi @mmarchetti. Sorry for delayed answer, I got so stuffs to do last month and so.
I'm not using pin32, why?
I'll do using #warning and check how it goes.

mmarchetti, 2018-01-23:
@mathgcosta, You mentioned a compile error - would you post it here? It would also be good if you can post a code section where you are using the Input or Output class.

mmarchetti, 2018-10-12:
Closing this for now, if more details become available we can reopen.

closed 2018-10-12 by mmarchetti