mmarchetti/DirectIO #25: Teensy 3.2/3.1 support

Synergyst, 2019-12-05:
Hello, 

I am curious about what would need to be done in order for me to add Teensy 3.2/3.1 support.
I have a Teensy 3.2 I can test with, which has the same underlying hardware as the 3.1 aside from the 3.1 being 3.3v-only.

Kind regards.

mmarchetti, 2019-12-06:
Hi,

Thanks for the question/interest! The main thing that is needed is a board definition file that maps Arduino pin numbers to ARM ports/pins.

The way these were created for most boards was to install the board support into Arduino, then run the `gen_pins.py` script. It reads the .h files for all the installed board types from Arduino, and creates the .h files needed by DirectIO. Presumably you would need to install something like Teensyduino (https://www.pjrc.com/teensy/td_download.html) in order for this to work. You probably need that anyway in order to compile and upload the code to the board.

Alternatively, you could create the .h file yourself based on the published pin mappings, and place it in the `include/boards/avr` directory.

The last step for the correct file to be used is to identify what compiler flag is used when targeting that board. From the [source,](https://github.com/PaulStoffregen/cores/blob/master/teensy3/Makefile#L56) it appears to be `TEENSYDUINO`. That has to be added like the other AVR boards here: https://github.com/mmarchetti/DirectIO/blob/master/include/ports_avr.h#L250 so the correct include file will automatically be selected for DirectIO to use.

Ping me if you have any questions/issues with getting this working.

mmarchetti, 2019-12-06:
Correction, that would likely need to be in one of the ARM board subdirectories depending on which architecture flag is set during compilation. That is referenced here: 

https://github.com/mmarchetti/DirectIO/blob/master/include/ports.h#L25-L34

Synergyst, 2019-12-06:
[image]

I'm using Arduino IDE 1.8.10 on Windows.
Am I doing something wrong perhaps?

mmarchetti, 2019-12-06:
To find the installed packages, it has to be run against the path where the installed libraries live, which should be `\Users\<username>\AppData\Local\Arduino18`.

mmarchetti, 2019-12-06:
OK, I tried this myself and haven't gotten `gen_pins.py` to work. Seems teensyduino doesn't install like the other board packages, and doesn't declare pins the same way in the header files. Your best bet might be to define the pin mapping manually. You just need to establish which pin numbers correspond to which port/pin on the microcontroller.

Synergyst, 2019-12-07:
I was testing the library on the Teensy 3.2 to check if DirectIO compiled against the Teensy 3.x from a program I made for another AVR before I opened the issue.
I don't need the direly need the library currently, but I'll probably get to mapping the ports, testing and then publishing any changes here if I end up really needing it for the Teensy 3.2/3.1.

It works great for the original AVR projects I needed and used it for though!