mmarchetti/DirectIO #26: PlatformIO Compiler  Error with direct_io_pin example

jakespeed1311, 2020-02-21:
Hello,
i tried to test the Lib with Platformio and compile the Examble "direct_io_pin.ino"
My version of PlatformIO ist  3.1.0 Core 4.2.1
I use an Arduino DUE an Programming Port tis ist Atmel SAM  4.1.0

 'reinterpret_cast<Pio*>(1074662912)' is not a constant expression
L
redifiniton of that thing ? But i don't kow how do fix this.
Thanks for help!

The error in the  Terminal Output:

> Executing task: C:\Users\admin\.platformio\penv\Scripts\platformio.exe run <

Processing due (platform: atmelsam; board: due; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/due.html
PLATFORM: Atmel SAM 4.1.0 > Arduino Due (Programming Port)
HARDWARE: AT91SAM3X8E 84MHz, 96KB RAM, 512KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduino-sam 1.6.12
 - framework-cmsis 1.40500.0 (4.5.0)
 - framework-cmsis-atmel 1.2.0
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting direct_io_pin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <DirectIO> 1.2.0
Building in release mode
Compiling .pio\build\due\src\direct_io_pin.ino.cpp.o
Archiving .pio\build\due\libdea\libDirectIO_ID6327.a
Compiling .pio\build\due\FrameworkArduinoVariant\variant.cpp.o
Compiling .pio\build\due\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\due\FrameworkArduino\Print.cpp.o
Compiling .pio\build\due\FrameworkArduino\Reset.cpp.o
Compiling .pio\build\due\FrameworkArduino\RingBuffer.cpp.o
Compiling .pio\build\due\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\due\FrameworkArduino\UARTClass.cpp.o
In file included from C:\Users\admin\.platformio\lib\DirectIO_ID6327/include/ports.h:28:0,
                 from C:\Users\admin\.platformio\lib\DirectIO_ID6327/DirectIO.h:23,
                 from C:/Users/admin/Documents/PlatformIO/Projects/200221-182940-due/src/direct_io_pin.ino:1:
C:\Users\admin\.platformio\packages\framework-arduino-sam\system\CMSIS\Device\ATMEL/sam3xa/include/sam3x8e.h:494:21: error: 'reinterpret_cast<Pio*>(1074662912)' is not a constant expression
 #define PIOA       ((Pio    *)0x400E0E00U) /**< \brief (PIOA      ) Base Address */
                    ~^~~~~~~~~~~~~~~~~~~~~~
C:\Users\admin\.platformio\lib\DirectIO_ID6327/include/ports_sam.h:29:36: note: in definition of macro '_define_port'
         static const u32 pio = u32(PIO); \
                                    ^~~
C:\Users\admin\.platformio\lib\DirectIO_ID6327/include/ports_sam.h:40:22: note: in expansion of macro 'PIOA'
 _define_port(PORT_A, PIOA);
                      ^~~~
C:\Users\admin\.platformio\packages\framework-arduino-sam\system\CMSIS\Device\ATMEL/sam3xa/include/sam3x8e.h:495:21: error: 'reinterpret_cast<Pio*>(1074663424)' is not a constant expression
 #define PIOB       ((Pio    *)0x400E1000U) /**< \brief (PIOB      ) Base Address */
                    ~^~~~~~~~~~~~~~~~~~~~~~
C:\Users\admin\.platformio\lib\DirectIO_ID6327/include/ports_sam.h:29:36: note: in definition of macro '_define_port'
         static const u32 pio = u32(PIO); \
                                    ^~~
C:\Users\admin\.platformio\lib\DirectIO_ID6327/include/ports_sam.h:44:22: note: in expansion of macro 'PIOB'
 _define_port(PORT_B, PIOB);
                      ^~~~
C:\Users\admin\.platformio\packages\framework-arduino-sam\system\CMSIS\Device\ATMEL/sam3xa/include/sam3x8e.h:496:21: error: 'reinterpret_cast<Pio*>(1074663936)' is not a constant expression
 #define PIOC       ((Pio    *)0x400E1200U) /**< \brief (PIOC      ) Base Address */
                    ~^~~~~~~~~~~~~~~~~~~~~~
C:\Users\admin\.platformio\lib\DirectIO_ID6327/include/ports_sam.h:29:36: note: in definition of macro '_define_port'
         static const u32 pio = u32(PIO); \
                                    ^~~
C:\Users\admin\.platformio\lib\DirectIO_ID6327/include/ports_sam.h:48:22: note: in expansion of macro 'PIOC'
 _define_port(PORT_C, PIOC);
                      ^~~~
C:\Users\admin\.platformio\packages\framework-arduino-sam\system\CMSIS\Device\ATMEL/sam3xa/include/sam3x8e.h:497:21: error: 'reinterpret_cast<Pio*>(1074664448)' is not a constant expression
 #define PIOD       ((Pio    *)0x400E1400U) /**< \brief (PIOD      ) Base Address */
                    ~^~~~~~~~~~~~~~~~~~~~~~
C:\Users\admin\.platformio\lib\DirectIO_ID6327/include/ports_sam.h:29:36: note: in definition of macro '_define_port'
         static const u32 pio = u32(PIO); \
                                    ^~~
C:\Users\admin\.platformio\lib\DirectIO_ID6327/include/ports_sam.h:52:22: note: in expansion of macro 'PIOD'
 _define_port(PORT_D, PIOD);
                      ^~~~
*** [.pio\build\due\src\direct_io_pin.ino.cpp.o] Error 1
========================================================================= [FAILED] Took 4.18 seconds =========================================================================

mmarchetti, 2020-02-21:
I haven't tested with PlatformIO. However, in the CMSIS header file (sam3x8e.h) there are two definitions of `PIOA`, depending on whether `__ASSEMBLY__` is defined.

When targeting AVR variants, DirectIO uses some `#define`/`#include` tricks to define `_SFR_ASM_COMPAT` without breaking the Arduino toolchain. Something similar may be needed here.

I won't be able to work on this for a bit. If you are looking for a workaround, you could try `#define __ASSEMBLY__ 1` and see if that works (or if it breaks something else in the toolchain, requiring similar trickery to the AVR configurations).

jakespeed1311, 2020-02-21:
hi,
ok thats bad news, tanks.
Jake

mmarchetti, 2020-02-24:
If you try the workaround, please let me know whether it worked for you.

jakespeed1311, 2020-02-25:
Hello,
no success, to be shure that i did notthing wrong here the code from direct_io_pin.ini examble.
I have seen some definitions, i am not able to fix this Problem by my own.

`
##Source Excamble ##

#define __ASSEMBLY__ 1
#include <DirectIO.h>

OutputPin pin(2);

void setup() {}

void loop() {
  while(1) {
    pin = HIGH;
    pin = LOW;
  }
}

#####################
`

Terminal Output
##################

Processing due (platform: atmelsam; board: due; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/due.html
PLATFORM: Atmel SAM 4.1.0 > Arduino Due (Programming Port)
HARDWARE: AT91SAM3X8E 84MHz, 96KB RAM, 512KB Flash
DEBUG: Current (atmel-ice) External (atmel-ice, blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduino-sam 1.6.12
 - framework-cmsis 1.40500.0 (4.5.0)
 - framework-cmsis-atmel 1.2.0
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting direct_io_pin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 7 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <DirectIO> 1.2.0
Building in release mode
Compiling .pio\build\due\src\direct_io_pin.ino.cpp.o
Archiving .pio\build\due\libFrameworkArduinoVariant.a
Compiling .pio\build\due\FrameworkArduino\USARTClass.cpp.o
Compiling .pio\build\due\FrameworkArduino\USB\CDC.cpp.o
Compiling .pio\build\due\FrameworkArduino\USB\PluggableUSB.cpp.o
In file included from C:\Users\admin\.platformio\lib\DirectIO_ID6327/include/ports.h:28:0,
                 from C:\Users\admin\.platformio\lib\DirectIO_ID6327/DirectIO.h:23,
                 from C:/Users/admin/Documents/PlatformIO/Projects/200224-203920-due/src/direct_io_pin.ino:2:
C:\Users\admin\.platformio\packages\framework-arduino-sam\system\CMSIS\Device\ATMEL/sam3xa/include/sam3x8e.h:494:21: error: 'reinterpret_cast<Pio*>(1074662912)' is not a constant expression
 #define PIOA       ((Pio    *)0x400E0E00U) /**< \brief (PIOA      ) Base Address */
                    ~^~~~~~~~~~~~~~~~~~~~~~
C:\Users\admin\.platformio\lib\DirectIO_ID6327/include/ports_sam.h:29:36: note: in definition of macro '_define_port'
         static const u32 pio = u32(PIO); \
                                    ^~~
C:\Users\admin\.platformio\lib\DirectIO_ID6327/include/ports_sam.h:40:22: note: in expansion of macro 'PIOA'
Compiling .pio\build\due\FrameworkArduino\USB\USBCore.cpp.o
 _define_port(PORT_A, PIOA);
                      ^~~~
Compiling .pio\build\due\FrameworkArduino\WInterrupts.c.o
C:\Users\admin\.platformio\packages\framework-arduino-sam\system\CMSIS\Device\ATMEL/sam3xa/include/sam3x8e.h:495:21: error: 'reinterpret_cast<Pio*>(1074663424)' is not a constant expression
 #define PIOB       ((Pio    *)0x400E1000U) /**< \brief (PIOB      ) Base Address */
                    ~^~~~~~~~~~~~~~Compiling .pio\build\due\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\due\FrameworkArduino\WString.cpp.o
~~~~~~~~
C:\Users\admin\.platformio\lib\DirectIO_ID6327/include/ports_sam.h:29:36: note: in definition of macro '_define_port'
         static const u32 pio = u32(PIO); \
                                    ^~~
C:\Users\admin\.platformio\lib\DirectIO_ID6327/include/ports_sam.h:44:22: note: in expansion of macro 'PIOB'
 _define_port(PORT_B, PIOB);
                      ^~~~
C:\Users\admin\.platformio\packages\framework-arduino-sam\system\CMSIS\Device\ATMEL/sam3xa/include/sam3x8e.h:496:21: error: 'reinterpret_cast<Pio*>(1074663936)' is not a constant expression
 #define PIOC       ((Pio    *)0x400E1200U) /**< \brief (PIOC      ) Base Address */
                    ~^~~~~~~~~~~~~~~~~~~~~~
C:\Users\admin\.platformio\lib\DirectIO_ID6327/include/ports_sam.h:29:36: note: in definition of macro '_define_port'
         static const u32 pio = u32(PIO); \
                                    ^~~
C:\Users\admin\.platformio\lib\DirectIO_ID6327/include/ports_sam.h:48:22: note: in expansion of macro 'PIOC'
 _define_port(PORT_C, PIOC);
                      ^~~~
C:\Users\admin\.platformio\packages\framework-arduino-sam\system\CMSIS\Device\ATMEL/sam3xa/include/sam3x8e.h:497:21: error: 'reinterpret_cast<Pio*>(1074664448)' is not a constant expression
 #define PIOD       ((Pio    *)0x400E1400U) /**< \brief (PIOD      ) Base Address */
                    ~^~~~~~~~~~~~~~~~~~~~~~
C:\Users\admin\.platformio\lib\DirectIO_ID6327/include/ports_sam.h:29:36: note: in definition of macro '_define_port'
         static const u32 pio = u32(PIO); \
                                    ^~~
C:\Users\admin\.platformio\lib\DirectIO_ID6327/include/ports_sam.h:52:22: note: in expansion of macro 'PIOD'
 _define_port(PORT_D, PIOD);
                      ^~~~
*** [.pio\build\due\src\direct_io_pin.ino.cpp.o] Error 1
C:\Users\admin\.platformio\packages\framework-arduino-sam\cores\arduino\USB\USBCore.cpp: In function 'void USB_ISR()':
C:\Users\admin\.platformio\packages\framework-arduino-sam\cores\arduino\USB\USBCore.cpp:688:21: warning: this 'else' clause does not guard... [-Wmisleading-indentation]       
                     else
                     ^~~~
C:\Users\admin\.platformio\packages\framework-arduino-sam\cores\arduino\USB\USBCore.cpp:690:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'else'
         UDD_Send8(EP0, 0);
         ^~~~~~~~~
========================================================================= [FAILED] Took 3.77 seconds =========================================================================
The terminal process terminated with exit code: 1

#

closed 2022-05-26 by jakespeed1311