mmarchetti/DirectIO #32: error: expected ';' before 'k'

simeon9696, 2021-04-13:
I'm trying to use OutputPin to declare a pin as output on runtime. But I get `error: expected ';' before 'k'` when trying to compile. It seems like a silly syntax error but I don't see how.  Below are my functions and logs

```
int getThmstAnlgRaw(int analogPin, u8 pwrPin){
  OutputPin(pwrPin) k;
  int reading = analogRead(analogPin);
  return reading;
}
```

```
The terminal process "C:\Users\simeon.ramjit\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

> Executing task: C:\Users\simeon.ramjit\.platformio\penv\Scripts\platformio.exe run <

Processing controllino_maxi_automation (platform: atmelavr; board: controllino_maxi_automation; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/controllino_maxi_automation.html
PLATFORM: Atmel AVR (3.0.0) > Controllino Maxi Automation
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <CONTROLLINO> 3.0.5
|   |-- <SPI> 1.0
|-- <DirectIO> 1.2.0
|-- <Blynk> 0.6.7
|   |-- <Ethernet> 2.0.0
|   |   |-- <SPI> 1.0
|-- <Ethernet> 2.0.0
|   |-- <SPI> 1.0
|-- <TimerInterrupt> 1.0.2
|-- <SPI> 1.0
Building in release mode
Compiling .pio\build\controllino_maxi_automation\src\main.cpp.o
src\main.cpp: In function 'int getThmstAnlgRaw(int, u8)':
src\main.cpp:166:21: error: expected ';' before 'k'
   OutputPin(pwrPin) k;
```
                     ^

mmarchetti, 2021-04-13:
Yes, the syntax for InputPin and OutputPin are different.

```
Output<3> pin3;
OutputPin pin4(4);
```

The first is C++ template syntax, the second is a normal (non-template) object constructor.

simeon9696, 2021-04-13:
You may need to update the documentation as it currently reads: 

```
void DoSomething(u8 pin)
{
    OutputPin(pin) my_output; // note pin number is now a constructor parameter
    my_output = HIGH;
}
```

mmarchetti, 2021-04-13:
Thanks for catching that!

closed 2021-04-13 by mmarchetti