mmarchetti/DirectIO #33: All port loop

jafal633, 2022-10-25:
Hi all

How to use this library ??

Want to control all port C pins on and off with loop and delay but I am not understanding how  with Arduino due
`   // Define a 4-bit port starting at port C2.
// This will control C2, C3, C4, C5 (pins 16-19).
OutputPort<PORT_C, 2, 4> my_port;

void setup() 
{
    my_port.setup();
}

void loop()
{
    // Turn on C2 (pin 16), and turn off the rest.
    my_port = 0x01;
}  `

Thanks in advance

mmarchetti, 2022-10-25:
The code you provided will set the port pins to the value you provided. Since it's 0x01, one pin will be set high and the others low. If you want dynamic behavior, you can provide different values in `loop()`, since `loop` is called repeatedly by the Arduino runtime. To implement delays, you can use Arduino timing functions such as `millis()` to determine when it is time to change values.

jafal633, 2022-10-26:
Hi Sir and thanks for answering

In this example for instance is it computable with Arduino due and how much pins I can  define ??

And without delay  the 10mhz  constant on all defined pins or dividing 

`       #include <DirectIO.h>

OutputPin pin(2);

void setup() {}

void loop() {
  while(1) {
    pin = HIGH;
    pin = LOW;
  }
}
      `
Thanks in advance

mmarchetti, 2022-10-26:
If you are using `OutputPort` on an ARM-based Arduino, you should be able to define ports up to 32 bits (assuming all 32 pins are available on the board).

I don't recommend using `OutputPin` if you know the pin number - it will not speed up your code. Use `Output<2> pin` instead.

If you want to implement delays, you can call the `millis()` function in your `loop()` to see whether it is time to take your next planned action, or call the Arduino `delay()` function in between your actions to create whatever delay you need. DirectIO doesn't help with those parts.

jafal633, 2022-10-26:
Thank you Sir